CopterTec/Marlin-Sovol_SV01 #1: Slow to nonn responsive media menu[BUG] (short description)

Gnjac0rp, 2020-01-04:
### Bug Description

### My Configurations

**Required:** Please include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

0n3-70uch, 2020-01-04:
Hi, it's included. Please check the release page. You can find the ZIP files there.

0n3-70uch, 2020-01-04:
Please try to reformat the sd card.

closed 2020-01-05 by CopterTec