CopterTec/Marlin-Sovol_SV01 #3: Neopixel/LED support

samd12, 2020-01-08:
If we could use a pin from the main board for control only with Neopixel or standard leds

CopterTec, 2020-01-09:
Hi, do you know which port can be used? I can compile a Neopixel version for you.

samd12, 2020-01-09:
I do not know the pin number but it is 2 rows above the bltouch pins, I believe it is pin 5 as per the linked video timestamp 12:20.   https://www.youtube.com/watch?v=LEq_HFO2LYs&t=639s     There is a signal, ground and positive for each row of pins.  
[image]

CopterTec, 2020-01-09:
Ok I will check it soon. Seems to be a nice feature.

samd12, 2020-01-09:
That sounds awesome.  Look forward to it.  For most people it needs to be configured for GRB not GRBW which most code comes with.

samd12, 2020-02-02:
You can close this.  I figured it out.

CopterTec, 2020-02-02:
Ok. You can write a short how to... Will help many people. 

I think it's bad to implement this directly into the firmware file. To many different pixel numbers.

closed 2020-02-02 by CopterTec