CopterTec/Marlin-Sovol_SV01 #4: [BUG] PlatformIO CLI compilation yields small .hex file

fghorow, 2020-04-07:
### Bug Description

Compiling your Marlin-Sovol_SV01 package yields drastically smaller .hex file size. Bug? Misconfiguration?

### My Configurations

[configs.zip](https://github.com/fghorow/Marlin-Sovol_SV01/blob/master/config/SOVOL/SV01/configs.zip)

### Steps to Reproduce

1. platformio run -e mega2560
2. Compare .pio/build/mega2560/firmware.hex filesize (148k) with your released version (Marlin_2.0.5.3-Sovol_SV01-1.6.3-en.hex) 459k.

**Expected behavior:** [What you expect to happen]

Filesizes to be the same.

**Actual behavior:** [What actually happens]
Filesizes differ enough for me to suspect that my .hex is not viable, or I would need to hack board hardware to install bootloader to recover from bad flash.

0n3-70uch, 2020-04-08:
Hi, I think you forgot the "_Bootscreen.h" and "_Statusscreen.h". Can you recheck it so that I can close this issue. Thx

CopterTec, 2020-04-08:
Hi, I think you forgot the "_Bootscreen.h" and "_Statusscreen.h". Can you recheck it so that I can close this issue. Thx

fghorow, 2020-04-08:
First off, THANKS for the response and for the suggestion! Your efforts are greatly appreciated as I struggle to print protective equipment for Covid-19 medical workers!

I'd be happy to re-check those files, but _how_? 

This is the first time I've even _heard_ of PlatformIO and tried to use it for making firmware.
I simply ran my command above in your checked-out master branch of your git repo, changed the number of mesh points in the manual mesh leveling from 3 to 5 to address a warped bed issue I'm having, and ran the `platformio` command above.

If there's a misconfiguration, it was through my complete ignorance of this build system. 

Is it possible for you to add the configuration for those `.h` files into the repo? If not, some brief documentation on how to do so at compile time would be helpful, I think.

Thanks!

CopterTec, 2020-04-08:
You are welcome ;-).

Some steps to compile the firmware:
1) Download the Soruce Code .zip file from the "Release" section
2) Unzip everything
3) Copy your needed config from /config/SOVOL/SV01 (example) to /Marlin
4) start PlatformIO (recommend VSCode)
5) click the compile button at the left bottom (in VSCode)

Thats all. Here is another auto build option:
https://marlinfw.org/docs/basics/install_platformio.html

fghorow, 2020-04-08:
Closing this issue because copying all configs from /config/SOVOL/SV01 (example) to /Marlin as suggested by @CopterTec successfully creates a bootable and flashable firmware.

Thanks!

closed 2020-04-08 by fghorow

CopterTec, 2020-04-08:
Great. Happy compiling.