CopterTec/Marlin-Sovol_SV01 #7: [FR] Change DEFAULT_MINSEGMENTTIME from 20000 to 50000

msteele999, 2020-06-19:
### Description

Would you consider changing #define DEFAULT_MINSEGMENTTIME 20000 to 50000 in the next release in Configuration_adv.h? 

I've been researching OctoPrint print quality issues and this setting looks to have some promise:

http://lokspace.eu/bad-print-quality-with-usb-or-octoprint-the-solution-is-here/?fbclid=IwAR22PHYObHijYrMj3DUGp9uc4oaMctQO2HE2CsDH6QO6zB3Tps78Ge7W_jM

Best regards,

Mark

CopterTec, 2020-06-19:
Hi, have you successfully tested this with your Sovol? Very interesting issue...

Greetings
Fabi

msteele999, 2020-06-19:
I have not tried compiling it myself (I'll be honest that I don't know how
to do that myself) but I have spoken to others that have and they indicate
it makes a huge difference.

I know this is outside the scope of your firmware, but do you know if there
is a guide for compiling your Marlin firmware with any changes I make to a
.hex file?

Best regards,

Mark

CopterTec, 2020-06-19:
Which version do you use? I can compile a test version next week...

msteele999, 2020-06-19:
2.0.5.3 English with BL Touch

Thank you so much!

I'll do some googling to see what it takes to setup my system to be able to
compile from source as well.

CopterTec, 2020-06-20:
Ok please try this version with
`#define DEFAULT_MINSEGMENTTIME 50000`

[TEST_Marlin_2.0.5.3-Sovol_SV01-1.6.3.1-BLTouch_en.zip](https://github.com/CopterTec/Marlin-Sovol_SV01/files/4807848/TEST_Marlin_2.0.5.3-Sovol_SV01-1.6.3.1-BLTouch_en.zip)

msteele999, 2020-06-20:
Awesome. I'll test today. I have a couple of long prints to do.

CopterTec, 2020-06-20:
Ok great. Have a nice weekend.

msteele999, 2020-06-21:
Loaded firmware successfully - testing some prints now.

CopterTec, 2020-06-22:
Ok... hope it will work better now.

msteele999, 2020-06-22:
Fabi,

This has made a positive impact on my prints when using OctoPrint -
primarily 1) less 'zits' and blobs, 2) less noticeable layer lines.

Thank you for doing this!

CopterTec, 2020-06-22:
Hi, ok great. It will be implemented in the next release:
https://github.com/CopterTec/Marlin-Sovol_SV01/commit/8855b9b664e6088c51fb03c88465ee869cefb0c1

closed 2020-06-22 by CopterTec

CopterTec, 2020-07-04:
Pre-Release 1.6.4 RC1 with implemented DEFAULT_MINSEGMENTTIME:
https://github.com/CopterTec/Marlin-Sovol_SV01/releases/tag/1.6.4_RC1

msteele999, 2020-07-04:
FYI. My board is the v2.2.1 silent board if that makes any difference.

CopterTec, 2020-07-06:
Ok. The sovol silent board?

msteele999, 2020-07-06:
Correct,

It's the creality ender5 silent board

https://www.amazon.com/dp/B085DM75Q6?ref=ppx_pop_mob_ap_share

CopterTec, 2020-07-06:
Ah now I notice that you are the same person ;-). Thanks.

msteele999, 2020-07-06:
Lol

chilman408, 2020-09-12:
hi guys, just wondering if the default_minsegmenttime change has a negative impact on printing from SD?

CopterTec, 2020-09-12:
Yes a bit curious...