CopterTec/Marlin-Sovol_SV01 #9: [FR] Question Only

msteele999, 2020-06-30:
I am replacing my factory main board with a V2.2.1 board (https://smile.amazon.com/gp/product/B085DM75Q6/ref=ox_sc_act_title_1?smid=A3SNVVMEXOSKIB&psc=1)

Have you heard of any issues running the current firmware on this board?

Thank you!

CopterTec, 2020-06-30:
Hi, that's a good question... Don't know if this firmware is running fine on the 2.2.1 board. Please give me a feedback.

msteele999, 2020-06-30:
OK - we'll find out soon

CopterTec, 2020-07-01:
Ok great. Thanks.

msteele999, 2020-07-03:
So - the printer powers on and everything looks OK with the new board - however it looks like the BL Touch commands do not work properly. When I attempt to home Z, the probe never drops so the nozzle crashes into the bed.

I've reloaded the firmware several times with the same results.

I loaded the Sovol version of the hex file and it acts the same way - so I assume this is a problem with the board and what it expects for the BL Touch commands

msteele999, 2020-07-03:
I found the issue - I plugged the BL Touch into D12 instead of D11 - continuing compatibility testing.

(sorry)

CopterTec, 2020-07-03:
Hi, ok.. :-)

msteele999, 2020-07-04:
Everything works perfectly. No issues to report.

CopterTec, 2020-07-04:
Perfect... good news. I will add this Board to the supported devices list.

CopterTec, 2020-07-04:
I changed the README.md
https://github.com/CopterTec/Marlin-Sovol_SV01/commit/0f96e9d58afe39272c04af267bd649f78deefe94

closed 2020-07-04 by CopterTec