CopterTec/Marlin-Sovol_SV01 #13: [FR] Question if the Touch Screen (Aux-1) port on SV01 User Guide is enabled?

chilman408, 2020-09-13:
I'm looking to swap out the OEM lcd with the TFT35 E3 by BTT.  Looks like it needs this port to enable the Touch Screen functions.

### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

CopterTec, 2020-09-14:
Hello, I think it's disabled by default. You need to compile your own firmware with a configuration for the touch screen.

closed 2020-09-14 by CopterTec

chilman408, 2020-09-14:
> Hello, I think it's disabled by default. You need to compile your own firmware with a configuration for the touch screen.

Thanks but I wouldn't know where to start...

CopterTec, 2020-09-14:
Please checkout the configuration.h in /configs/SOVOL

chilman408, 2020-09-14:
Thanks for the pointer!

I reviewed the configuration.h file against what is available for the CR10S and the Ender 3.  

I don't see any difference between what is enabled in the Creality FW versions to what is enabled in the SV01 firmware in terms of reference to AUX1, Serial Port, Touch Screen, Display, or TFT...  (I just search these terms)

Can I assume that this is enabled if the Serial Port 0 is enabled?  

(Completely new at this...)

CopterTec, 2020-09-14:
Did you checked the configuration_adv.h too?

chilman408, 2020-09-14:
Yes, I actually did more research on this since last posted.

It seems for the Creality V2.2 board, the AUX1 and USB traces are routed together, so the AUX1 port should work if the USB is working.  

However, this creates a problem where only one of these port may be used at any time.  

I'm not sure in what scenario I would use the USB port besides for FW flash or to hook up an Octopi.  For FW flash, I can easily just disconnect the TFT35 and it seems (from BTT's github for the TFT35), Octopi may be connected to the TFT35 directly.

There is a lot of information here that I need to think about.

Thanks!

CopterTec, 2020-09-14:
Ah ok... so better you use another board with 32bit microcontroller. Much more options.

chilman408, 2020-09-14:
Compiling a working firmware for a 32bit board would be the difficult part for me.

It would be great if you can select a good 32bit board, pre-compile as part of the Marlin releases?

CopterTec, 2020-09-15:
Oh thats impossible for us... we sold our Sovol printer some month ago.