CopterTec/Marlin-Sovol_SV01 #14: [FR] Supporting Linear Advance with K factor 0?

chilman408, 2020-09-23:
### Description

Hi, wondering if in the next release you will support linear advance with K=0?  This way, people who want to mess with it can mess it and people who don't, it is effectively disabled.

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

CopterTec, 2020-09-23:
Hello, nice idea. Is it 100% disabled with k=0?

msteele999, 2020-09-23:
I believe that is correct - K=0 disables LA. I would vote for this as well.
I'd like to do some extra fine tuning so I can move to PrusaSlicer or Cura
4.7.1 - but I need that M900 working :-)

CopterTec, 2020-09-23:
Ok I will add this option with the next release.

CopterTec, 2020-09-23:
Welcome

msteele999, 2020-09-25:
I'll be happy to beta test if you want.

Best regards

CopterTec, 2020-09-25:
Sorry no time at the moment... first we need to upgrade marlin.

msteele999, 2020-09-25:
By all means!