CopterTec/Marlin-Sovol_SV01 #16: [FR] Touch Display support

CopterTec, 2020-10-16:
Support for the original Sovol touch display.

HotwiresFPV, 2020-10-25:
When will this be available?

CopterTec, 2020-10-25:
It is only possible if Sovol publishes the source code.

HotwiresFPV, 2020-10-28:
> It is only possible if Sovol publishes the source code.

Sovol just released the source code.

CopterTec, 2020-10-28:
It would also possible by try and error. But two problems: we need the hardware and time.

HotwiresFPV, 2020-10-28:
> It would also possible by try and error. But two problems: we need the hardware and time.

I can test

CopterTec, 2020-10-29:
You can download the source code and test different settings... We don't have time for that at the moment. Sorry.

HotwiresFPV, 2020-11-23:
> You can download the source code and test different settings... We don't have time for that at the moment. Sorry.

Compiled the firmware with the source code and not working properly. I ju

CopterTec, 2020-11-23:
Hello, did you copy the config files from /config/Sovol to /Marlin?

HotwiresFPV, 2020-11-23:
Yes. I did

CopterTec, 2020-11-23:
Ok. Please reinstall VSCode and Platformio... Most of the time it's a library problem.

HotwiresFPV, 2020-11-23:
I installed VScode and Platform IO on 2 different computers Mac and Windows
same results using coptertec source code and Sovol source code.

CopterTec, 2020-11-23:
You did something wrong... but i don't know what. This is the wrong page for compiler issues. Please check the Platformio or Marlin documentation.

HotwiresFPV, 2020-11-23:
With coptertec 1.6.3 source code and config I can compile fine. It has
something to do with sovol source code.

CopterTec, 2020-11-23:
Yeah they change to much stuff in an unprofessional way of coding. But this page is only for CopterTec firmware issues. Not for general Marlin issues or Sovol bugs. So please contact Sovol directly ;-).