CopterTec/Marlin-Sovol_SV01 #18: [FR] Which Configuration.h / Configuration_adv.h do I change?

msteele999, 2020-10-28:
I am changing my hotend to an all metal E3D V6 tonight and need to change the max temp and the thermister type. I noticed there are several Configuration.h and Configuration_adv.h files in the package - do I change the one in the Marlin folder or the one in the Config/Sovol folders?

I'm not that versed with how the configuration files work in the package.

Best regards,

Mark
### Description

### Feature Workflow

1. [First Action]
2. [Second Action]
3. [and so on...]

#### Additional Information

* Provide pictures or links that demonstrate a similar feature or concept.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

msteele999, 2020-10-28:
I think I found the answer in your message threads on the main page:

For my version, you copied the Config/Sovol/myprinterversion to the /Marlin directory already - so I would modify the configuration files in the /Marlin directory for the changes to go into effect.

Is that correct?

msteele999, 2020-10-28:
Hmmm - nope - that does not seem right.

OK - so - do *I* copy over the /Config/Sovol/myprinterversion to the /Marlin directory and then compile? I'm really sorry about all the questions - when I flashed my printer with your hex code - I didn't have to think about all this :-)

CopterTec, 2020-10-29:
Hello, first of all. This site is for issues not for support. Now back to your question:
Just copy and replace all files from /Config/Sovol/myprinterversion to /Marlin. Thats all.

closed 2020-10-29 by CopterTec

msteele999, 2020-10-29:
Thank you. I was successful.

I am profoundly sorry for posting that question there. I won't let it
happen again.

Best regards,

Mark

CopterTec, 2020-10-29:
Nice to hear that you were successful.

No problem... you can write but use the "edit" function please... less emails ;-).

Have a nice day
Fabi