CopterTec/Marlin-Sovol_SV01 #20: 1.6.3 source not correct?

Boundless-Code, 2020-12-23:
I downloaded the source files for 1.6.3. New to editing marlin, but when I look though configuration.h it seems like this is not what you used to compile. Example, acceleration and travel are set to 3000 in the source, but not in the complied version. Am I missing something or is this not the ready to compile source?

I just wanted to poke around and maybe make some small changes to an already good to go file.

CopterTec, 2020-12-23:
Hello, as we wrote in the comments: you need to copy the files from /config/Sovol/Type to /Marlin. Greetings

closed 2020-12-23 by CopterTec

Boundless-Code, 2020-12-23:
> 
> 
> Hello, as we wrote in the comments: you need to copy the files from /config/Sovol/Type to /Marlin. Greetings

Not sure where that information was exactly. Got it now though, thank you!

CopterTec, 2020-12-23:
You are welcome ;-). 
Checkout this comment section for many issues:
https://www.coptertec.de/news/2019/marlin-2-0-for-sovol-sv01/

Boundless-Code, 2020-12-24:
> 
> 
> You are welcome ;-).
> Checkout this comment section for many issues:
> https://www.coptertec.de/news/2019/marlin-2-0-for-sovol-sv01/

Appreciate you creating and maintaining this firmware version for the SV01. Thank you!