CopterTec/Marlin-Sovol_SV01 #21: [FR] Support for Sovol SV03

CopterTec, 2021-02-19:
Thanks to Chris for his help. Parameters for the next release:

```
#define CUSTOM_MACHINE_NAME "SOVOL SV03"

// The size of the print bed
#define X_BED_SIZE 350
#define Y_BED_SIZE 350

// Travel limits (mm) after homing, corresponding to endstop positions.
#define X_MIN_POS 0
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 390
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 410
```

Nerone517, 2021-11-05:
Guten Tag, ist es die Firmware des SV01 auf den SV03 umzuwandeln ?

CopterTec, 2021-11-05:
> Guten Tag, ist es die Firmware des SV01 auf den SV03 umzuwandeln ?

Hallo, mit etwas Arbeit können Sie sich das umarbeiten. Einfach den Source Code herunter laden und anpassen.

Nerone517, 2021-11-07:
Leider kann ich das nicht, ich weiß zu wenig über Marlin

CopterTec, 2021-11-08:
> Leider kann ich das nicht, ich weiß zu wenig über Marlin

Hallo, wir haben den Drucker gegen einen Prusa ersetzt da der Support von Sovol nicht ausreichend war. Daher können wir das leider nicht mehr machen.