CopterTec/Marlin-Sovol_SV01 #23: [BUG] only heating, does not start printing... ?

ozett, 2021-05-23:
### Bug Description

i found your firmware for my new sovol s01. i printed the cube with all defaults successful, than i updated the firmware
with cura to you4 1.6.3 and after 4minutes waiting with E:heating nothing happens now...
is this normal? how long do i have to wait? in stock settings it printed after reaching the temp.

https://github.com/CopterTec/Marlin-Sovol_SV01/releases/download/1.6.3/Marlin_2.0.5.3-Sovol_SV01-1.6.3-en.hex

**Expected behavior:** [What you expect to happen]

start printing after reaching temp

**Actual behavior:** [What actually happens]

reaching temp. still showing heating....

#### Additional Information

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ozett, 2021-05-23:
uhhh i found this: "Heats up but does not print = calibrate PID"..
now looking on your site how i do this calibration..

ozett, 2021-05-23:
my luck, i am not the only idiot out here.. 😄 
https://www.coptertec.de/news/2019/marlin-2-0-for-sovol-sv01/?replytocom=1612

must be sowhere in the LCD menue... 
at the moment after downgrade teh SOVOL is printing...must wait for the end to try to find it...

ozett, 2021-05-23:
no luck, at all.
i found in the LCD../configuration/auto-temp: off
switched to "on", stored all on SD-Card.

but no change.

do i miss something?

CopterTec, 2021-05-25:
Hello, this is a common issue. Please run the nozzle PID autotune. Have a nice day.

closed 2021-05-25 by CopterTec

ozett, 2021-05-25:
sorry, but i am stuck.

is it a menu-option on the LCD?

i tried to find (like others in the thread).
maybe a hint, where to look?
or link to the marlin-lcd-menue structure?
or a picture? 😢  as a little help?

or do i have to do it with 3rd party-software? 
I am a little lost. the linked video has also no real instruction for that.

would appreciate it (and your firmware has some more option, looks pretty good,) ..like your copters
😸 

kind regards,
toz

CopterTec, 2021-05-25:
Yes it's an option in the menu. Here is the menu tree:
https://marlinfw.org/docs/features/lcd_menu.html

Check out the "Tune" section. Greetings

ozett, 2021-05-25:
i will quick jump to my sovol and do the magic.
💐 thanks for the quicker than zero-latency response.
this shades light of hope for me.

🤝

CopterTec, 2021-05-25:
You are welcome ;-)

ozett, 2021-05-25:
do it have to do autotune trough sending commands like this:

[image]

CopterTec, 2021-05-25:
You can use the commands but there must be an option in the menu tree (in the deep of this menu). Sorry we sold our Sovol last year... so I can't check it now.

ozett, 2021-05-25:
did not found on the printer-menue:

[image]

i send command with cura-console:
[image]

CopterTec, 2021-05-25:
There it is... Temperature > PID Autotune E1 (Extruder 1). Put in your most used temperature (for example 210°C).

ozett, 2021-05-25:
ok. will try.
(cura command did not help)

ozett, 2021-05-25:
for reference: change this value
[image]

it seems that i dont have strictly enable "Autotemp: ON" higher in this menu to get this working.
my printer not starts after changing the value from 150 to 220 (for my PLA).

thousand thanks for all
💐

CopterTec, 2021-05-25:
Thanks for your feedback. Happy printing.